NatLabRockies/ssc #601: `batt_bank_replacement` indexed differently from `batt_replacement_schedule_percent`

dguittet, 2021-08-12:
**Describe the bug**
`batt_bank_replacement` is length `analysis period` + 1 whereas `batt_replacement_schedule_percent` is length `analysis_period`

from cmod_singleowner.cpp:
```
cf.at(CF_battery_replacement_cost, i + 1) = batt_rep[i+1] * replacement_percent[i] * 0.01 *
                    cf.at(CF_battery_replacement_cost_schedule, i + 1);
```

**Expected behavior**
They should be indexed similarly. This inconsistently will lead to input errors.

dguittet, 2021-08-19:
@brtietz Thanks for the context. I'm not sure how many other technology cmods report analysis_years + 1 length output arrays. I'll add it to discussion agenda for next week.

brtietz, 2021-08-24:
SAM meeting 8/24: Paul found some similar annual arrays in Geothermal

PR #641 (2021-11-05): Ssc 601 consistent batt lengths
Fixes #601 - as well as other variables with similar length issues.

All inputs to financial models from technology models should now be length analysis_period. These are then lengthened after they are use to analysis_period + 1 to include year zero.

This includes a new test file (vartab_test.h/cpp) so re-run CMake prior to testing.
closed 2021-11-05 by brtietz