NatLabRockies/ssc #644: cf_ptc_total doesn't work

dguittet, 2021-11-06:
**Describe the bug**
I assume `cf_ptc_total` = `cf_ptc_fed` + `cf_ptc_sta` but the total seems to be 0 even when fed is non-zero.

**To Reproduce**
Steps to reproduce the behavior:
1. Open default Wind Single Owner case (Federal PTC will be active)
2. Click on 'Simulate'
3. Try out this script: 
[image]

cpaulgilman, 2021-11-08:
I think `cf_ptc_total` should be removed because the state PTC and federal PTC are treated separately in the cash flow: The the state PTC is taxable income for Federal tax purposes. For example, see [here](https://github.com/NREL/ssc/blob/9cf27a1bb29f9488eb1b13f95a3ce35315d3ea9b/ssc/cmod_singleowner.cpp#L2721) for Single Owner. So the total PTC (sum of State and Federal) does not really have significance for cash flow calculations.

The total PTC is only calculated by cmod_cashloan, but not used to calculate any results. It is assigned as an output in common,cpp, but not assigned a value in any of the financial model compute modules.

Solution is to remove `cf_ptc_total` from common,cpp and cmod_cashloan.cpp.

LK Script for testing with Wind / Single Owner:

```
a = get('cf_ptc_fed');
b = get('cf_ptc_sta');
c = get('cf_ptc_total');
d = get('itc_total'); // not relevant for this conversation

outln(a);
outln(b);
outln(c);
outln(d);
```

dguittet, 2021-11-13:
@cpaulgilman Thanks for the explanation!

PR #1072 (2023-10-27): Remove total PTC output from all models
Fixes #644

Remove cf_ptc_total from cashloan and common. 

Ignoring legacy cmod_ippppa.
closed 2023-10-27 by cpaulgilman