NatLabRockies/ssc #666: FOM automated dispatch uses ppa_price_input even if irr target is selected

brtietz, 2021-11-18:
The front of meter automated dispatch methods will use the ppa_price_input value from the defaults/GUI even if the financial models are run in IRR target mode:

[image]

The above code needs to reference ppa_soln_mode and come up with a surrogate PPA value, or the battery GUI pages need to warn the user that the value will be used.

mjprilliman, 2021-11-18:
Steps to see issue: 
1. Open Standalone Battery - Single Owner case
2. On Revenue page, Select 'Specify PPA price', change PPA price to 0, then select 'Specify IRR target'
3. On Dispatch page, select Automated dispatch
4. Simulate
5. Results have NaN battery energy charge and roundtrip efficiency

[Automated_dispatch_PPA.zip](https://github.com/NREL/ssc/files/7566799/Automated_dispatch_PPA.zip)

brtietz, 2022-04-12:
Fixed with the above pull requests

closed 2022-04-12 by brtietz