NatLabRockies/ssc #675: CMPvsamv1BatteryIntegration_cmod_pvsamv1.ResidentialDCBatteryModelPriceSignalDispatch faling on Travis and runs fine on Windows

sjanzou, 2021-11-29:
The one failing test runs fine on Windows
[image]

Retesting on Travis running single test     ::testing::GTEST_FLAG(filter) = "CMPvsamv1BatteryIntegration_cmod_pvsamv1.ResidentialDCBatteryModelPriceSignalDispatch";
[image]

batt_DOD_cycle_average output difference - separate issue
27.29 on Windows 
27.52 on Travis

_Originally posted by @sjanzou in https://github.com/NREL/ssc/issues/673#issuecomment-981348314_

dguittet, 2021-11-29:
It passes on Mac also with value 27.290548

brtietz, 2021-11-29:
This appears to be a recurrence of issue https://github.com/NREL/ssc/issues/614 - the linux and windows implementations end up with slightly different dispatch patterns. 

With the defaults currently in the GUI, windows schedules dispatch for step 966 but not 967, whereas Linux does the reverse. This causes the SOC to diverge, with greater divergence in the dispatch plan from there. It amounts to about 4 out of 926 kWh in year 1.

Interestingly, the Linux test is displaying the old value: https://github.com/NREL/ssc/commit/442ea5b6a787f75335aece7199efb6f442f943d3 Unfortunately due to the Travis outages I can't see if that commit caused the test failure or if the error was introduced a few days earlier and that's when it was exposed.

Would it be acceptable to increase the tolerance on the test for now and clean this up after the release? I created a branch addtl_lifetime_testing_in_integration_tests that shows that this issue is restricted to price signals dispatch - we could merge that in for additional test coverage of cycling differences between platforms.

brtietz, 2021-11-29:
I checked out 314cb53e0 (from Nov 17th, before the anti-reflective glass change) and the test passes there. This means that the dispatch changes in https://github.com/NREL/ssc/pull/662 exposed the Linux failure and my fix to https://github.com/NREL/ssc/issues/614 is insufficient.

I think this is a strong argument for a tolerance increase for this release, and to re-open https://github.com/NREL/ssc/issues/614 for a more robust solution in the future.

sjanzou, 2021-11-29:
That seems like a reasonable solution for the release…

closed 2021-11-30 by sjanzou