NatLabRockies/ssc #698: Battery NMC Lifetime Model `Q_Li` is wrongly initialized

yzhang318, 2021-12-28:
**Describe the bug**
In L121 of `shared/lib_battery_lifetime_nmc.cpp` in function `lifetime_nmc_t::runQli`:

```CPP
double dQLi1dt = 0;
    if (state->nmc_li_neg->dq_relative_li1 == 0) {
        if (state->day_age_of_battery > 0)
            dQLi1dt = b1 / sqrt(state->day_age_of_battery);   <--- L121
    }
    else
        dQLi1dt = 0.5 * b1 * b1 / state->nmc_li_neg->dq_relative_li1;  <--- L124
    double dQLi2dt = b2 * (double)dn_cycles;
    double dQLi3dt = fmax(0.0, b3 - state->nmc_li_neg->dq_relative_li3) / tau_b3;

    state->nmc_li_neg->dq_relative_li1 += dQLi1dt;
```

As according to the [cited paper](https://ieeexplore.ieee.org/document/7963578) (p3, left column) the equation (4) is:

[image]

I think the marked line should be to initialize `dQLi1dt`, the derivative of `Q_Li1`:

[image]

The above equation fits L124 and a factor of `0.5` is missing versus L121. If so, L121 should be: `0.5 * b1 / sqrt(state->day_age_of_battery)`.

dguittet, 2022-01-06:
These equations are pulled from a newer version of the model where some changes to the coefficients and structure were made. Double checking with that model, I see the initialization in line 121 matches with what is currently in SSC.

For the initialization when dQLi1 == 0 and age > 0, dQLi1 = b1*t^0.5, and dQLi1dt = dQLi1/dt_days_life = dQLi1/1.

yzhang318, 2022-01-14:
@dguittet Thanks! I ignored that when t = 0, the derivative of sqrt(t) is infinite. I agree with your opinion. A rough estimation goes as:

```C
// day 1, using L121
Q_Li1 = b1 / 1 = b1

// day 2, L124
Q_Li1 = Q_li1(day1) + 0.5 * b1 * b1 / Q_li1(day1) = b1 + 0.5 * b1 = 1.5 * b1

// day 3, L124
Q_Li1 = 1.5 * b1 + 0.5 * b1 * b1 / 1.5 / b1 = 1.83 * b1

// day 4, L124
Q_Li1 = 1.83 * b1 + 0.5 * b1 * b1 / 1.83 / b1 = 2.10 * b1

// if directly using accumulated func at day 4:
Q_Li1 = b1 * sqrt(4) = 2 * b1
```

The original code behaves correctly and yields correct numbers if initialized from day 1 as shown above. 

I think this issue can be closed. Thanks again for your reply.

closed 2022-01-14 by dguittet