NatLabRockies/ssc #720: Electricity peak demand charges produce negative numbers, and don't correlate with peak from grid output

brtietz, 2022-01-11:
@brtietz I'm finally getting around to documenting these changes for Help. 

The default PVWatts/Commercial case has TOU demand rates with 2 TOU periods each and 2 tiers, so I was expecting to see 4 values per month in the hourly "Demand charge with system" output. But instead, I'm seeing a handful of "Demand charge with system" in $ values per month that do not always correspond to the "Electricity peak from grid" in kW, and I'm seeing negative values in the first day of the month even without billing demand ratchets. Can you help explain what these values represent?

[image]

_Originally posted by @cpaulgilman in https://github.com/NREL/ssc/issues/652#issuecomment-1009426244_

cpaulgilman, 2022-01-28:
@brtietz, you replied to my [comment in #652](https://github.com/NREL/ssc/pull/652#issuecomment-1009426244):

> The intention with these is to record the incremental demand charge, so each time the demand charge increases it would produce a line proportional to the increase in the "demand charge with system" output. This is to assist with the contribution of grid charging to the LCOE and LCOS for the battery.
>
> The negative lines are definitely a bug. I'm also not sure why there would be a blue line without an orange line, so it looks like there might be two bugs here.

With this fix, there is now one kW peak per TOU period per month (orange line). In this case (default PVWatts/Commercial), the peaks happen to be at the beginning of the month in January, so I've zoomed the graph in to the first couple of days. I also used stepped lines to make it easier to see the time steps:

[image]

There are multiple demand "charges" per TOU period (blue). (The monthly demand charge per TOU period is the sum of these demand charges by period.) I don't understand why there are demand charges time steps (blue) with no peak demand (orange). 

Looking at electricity to/from grid shows that SAM is recording a demand charge as the simulation proceeds for the current time step if electricity from grid is the maximum so far, but that makes it seem like the final demand charge amount on the bill includes these cumulative peaks instead of just the peaks over the month per TOU period:

[image]

In this case, the rate structure has two TOU periods with two tiers per TOU period:

[image]

brtietz, 2022-01-28:
In the implementation in this PR, the blue lines are the incremental charges, as discussed. Only the final peak for each TOU period is represented in orange. The help describes this as:

[image]

Would you prefer the output to record the kW for each incremental charge?

cpaulgilman, 2022-01-28:
> Would you prefer the output to record the kW for each incremental charge?

No. I'm just trying to understand why the monthly demand charge is calculated from the incremental charges. If the hourly demand charge is just for internal use by the LCOS calculations then I think it may be confusing to include it as an output, or it should be relabeled as "Incremental demand charge with system for LCOS calculation." For the purpose of understanding the electricity bill calculation, I think users would expect to see a single dollar amount associated with each kW peak.

brtietz, 2022-01-28:
This is a separate calculation. The blue line comes from here: https://github.com/NREL/ssc/blob/3d5a0b418957f96ed55e1afcf303d013c0b10d1e/ssc/cmod_utilityrate5.cpp#L1691

The final demand charge is applied to payment here: https://github.com/NREL/ssc/blob/3d5a0b418957f96ed55e1afcf303d013c0b10d1e/ssc/cmod_utilityrate5.cpp#L1946

(with analagous functions for ur_calc_timestep)

I'm fine with the name "Incremental demand charge with system for LCOS calculation." - this would also be a potentially insightful output for price signals dispatch.

brtietz, 2022-02-11:
Fixed in https://github.com/NREL/ssc/pull/733

PR #733 (2022-01-31): Fix SSC 720
- Initializes the demand charge used by year1_hourly_dc_with_system to zero to prevent negative values
- Only sets values in year1_hourly_dc_peak_per_period once each month, preventing redundant values. Should now be one value per TOU period

Closes #720
closed 2022-02-11 by brtietz