NatLabRockies/ssc #917: Check NOCT model implementation

mjprilliman, 2022-10-31:
**Describe the bug**
A question on the implementation of the NOCT cell temeprature model equation used with the CEC performance models was presented in https://www.researchgate.net/publication/361463601_PV_Module_Operating_Temperature_Model_Equivalence_and_Parameter_Translation. Evaluate the implementation of the model based on the definitions presented in the SAM technical reference, the DeSoto model paper, and the Duffie and Beckman textbook. Confirm whether the POA irradiance used in the model should be before or after reflection effects, and consider how this definition effects the efficiency and absorptance-transmittance product variables used in the model.

kanderso-nrel, 2022-12-12:
Perhaps the same thing on the pvlib side: https://github.com/pvlib/pvlib-python/issues/1462

adriesse, 2023-12-11:
Any thoughts or new insights on this?

mjprilliman, 2023-12-12:
> Any thoughts or new insights on this?

Hi Anton,

Thanks for bringing this back to attention. I had previously investigated this issue and compared SAM's implementation with the referenced equations in Duffie and Beckman 3rd edition. At that time I felt that SAM's implementation was consistent with Duffie and Beckman, but there was a bit of uncertainty in how to define the irradiance term. I did mean to follow up with you at that time but this issue got lost in the shuffle a bit. I will take a look at this again soon and compile my notes. Perhaps we could find a time to discuss this further in early January 2024?

adriesse, 2024-02-01:
Here's the relevant comment from my paper for convenience.

> During a PV system simulation the value of Gpoa is typically reduced by the incidence angle modifier (IAM) at each time step so that module electrical output and thermal input are both reduced simultaneously for reflection losses. SAM takes a different approach and reduces the transmittance-absorptance product (τα) in the temperature model at higher incidence angles. This causes two problems: first, the electrical output in the thermal balance is not reduced; and second, since SAM equates (τα) and (τα)noct, the implicit empirical parameter Uo is also changed at every time step. It seems likely that this is unintended behavior of the SAM software, and if so, it can be corrected easily by subtracting reflection losses from Gpoa prior to running the temperature model.

In D&B, (τα) is constant, right?

mjprilliman, 2024-02-02:
> Here's the relevant comment from my paper for convenience.
>
>
> In D&B, (τα) is constant, right?

Correct, it is an assumed constant for the system energy balance. For the NOCT balance, going from 23.3.2 to 23.3.3 (which is the ratio of 23.3.1 to 23.3.2 changes τα * G_T,NOCT to just G_NOCT, which is just a constant 800 W/m2 regardless of τα, so it's not immediately clear to me how the τα for the NOCT energy balance factors into things.

adriesse, 2024-02-02:
I'm not sure what to suggest.  Maybe work through some cases by hand or in Excel (gasp)?

adriesse, 2024-09-15:
Just passing by...  :)

mjprilliman, 2024-09-20:
> Just passing by... :)

Hi Anton, thanks for bringing this back to attention. I can hopefully take a look at this issue again in October. The relevant code is here: 

https://github.com/NREL/ssc/blob/274560a9eba141154f7dd52a8d3e0fe5690b3b84/shared/lib_cec6par.cpp#L216-L277

If you have a suggested edit of the equation based on how it is written here, that would be much appreciated.

PR #1233 (2024-11-04): Noct taualpha
-Use Geff_total in NOCT model rather than G_total, remove irradiance scaling of tau*alpha product
-Update tests
-Fixes #917
closed 2024-11-04 by mjprilliman