NatLabRockies/ssc #1006: All technology outputs for financial models to be analysis period long or single values

sjanzou, 2023-03-11:
The fuel cell model has fueldcell_dispatched output of length analysis period + 1 and is used for production o and m costs in financial models.

The convention for all technology model outputs that are used in financial models is:
1. analysis period for lifetime mode
2. single values for single year technology compute modules

See discussion at https://github.com/NREL/SAM/issues/659#issuecomment-1464105549 for more details

janinefreeman, 2023-09-15:
@sjanzou has this been fixed already in the hybrids work, by chance?

sjanzou, 2023-09-17:
> @sjanzou has this been fixed already in the hybrids work, by chance?

Not yet fixed in hybrid branch 
[image]

closed 2023-11-08 by sjanzou