NatLabRockies/ssc #1015: Check that load and critical load data have same time step

cpaulgilman, 2023-04-19:
**Describe the bug**

SAM UI checks added in https://github.com/NREL/SAM/pull/1390 display UI error messages when critical load and load data have different time steps, but SSC does not prevent simulation from running under this condition.

**To Reproduce**

1. Using a build of SAM from the Patch branch that includes https://github.com/NREL/SAM/pull/1390, open the SAM file and load the 15-minute weather file from [here](https://github.com/NREL/SAM/files/11275249/crit-load-timestep-test.zip).
2. Go to the Grid Outage and Electric Load pages and see the error messages.
3. Run a simulation and note no simulation message. 

On Data Tables tab of Results page, note that "Critical load in this timestep" and "Lifetime electricity load" are of correct length: Hourly critical load data is converted to 15-minute data.

Critical load data 60-minute input:
[image]

Resulting output, with 60-minute critical load data converted to 15-minute time steps:
[image]

**Expected behavior**
I would expect a simulation message if time steps do not match, but perhaps the automatic conversion is the behavior we intended?

brtietz, 2023-05-03:
@cpaulgilman I created an SSC branch that should fix this (ssc_1015_errors_for_load_mismatch), however when using the test file above the errors are not triggered. The issue seems to be that while `load_user_data` has 15 minute data, `load` is still being passed to SSC as an 8760:

[image]

Do you have time to debug the callbacks and equations to see why that might be happening? I'll create a pull request once we get the expected behavior with the test file.

cpaulgilman, 2023-05-04:
Yes, I'll look into the SAM UI side of things.

cpaulgilman, 2023-05-05:
Callbacks should be fixed in https://github.com/NREL/SAM/tree/load_ui_checks.

closed 2023-05-05 by cpaulgilman