NatLabRockies/ssc #1037: Input checks for pvsamv1 result in inputs required for disabled subarrays

cpaulgilman, 2023-07-13:
Simulation error when `subarray2_track_mode` is not assigned even when `subarray2_enable = 0`:

```
Error: check fail: reason unassigned referenced, with 'subarray2_track_mode' for: subarray2_monthly_tilt time -1.000000
EXEC_FAIL: pvsamv1
```
To replicate, run this script in SDKTool with SSC Version 280:

[pv-subarray-input-tests.zip](https://github.com/NREL/ssc/files/12045024/pv-subarray-input-tests.zip)

See https://github.com/NREL/pysam/issues/155

caseyzak24, 2023-09-29:
@janinefreeman I assume this will be resolved in the next release. Do you have a sense of when that will be? Thanks!

janinefreeman, 2023-10-02:
Hi @caseyzak24 ! We are hoping for a release in mid-November.

cpaulgilman, 2023-10-02:
The issue here is that `subarray2_track_mode` is required for some input variable checks even when `subarray2_enable` is set to zero. 

For example, in order to check that `subarray2_tilt` is not set for the 2-axis or seasonal tilt options, the [line in the var table](https://github.com/NREL/ssc/blob/e12327536bed04258632ca66ea3b5aec6118d2ef/ssc/cmod_pvsamv1.cpp#L165C1-L165C498) for `subarray2_track_mode` needs to check `subarray2_track_mode`:

```
{SSC_INPUT, SSC_NUMBER, "subarray2_tilt", "Sub-array 2 Tilt", "degrees", "0=horizontal,90=vertical", "System Design", "subarray2_enable=1&subarray2_track_mode~2&subarray2_track_mode~4", "MIN=0,MAX=90", "" },
```
In other words, for this variable check to pass the following has to be true:

```
subarray2_enable == 1 && subarray2_track_mode != 2 && subarray2_track_mode != 4
```
So, although `subarray2_track_mode` is not required for the simulation, it is required for the variable check logic. Given the way; the variable table is set up, I'm not sure there is a way to avoid this, but I will investigate further.

sjanzou, 2023-10-03:
Note, that in compute_module::check_required the Boolean expressions are evaluated from left to right. Therefore, if the first constraint fails, the remaining constraints are not checked.  
[image]

The example file has a failure for “subarray2_monthly_tilt” because only subarray_track_mode is checked and not subarray2_enable=1 first
[image]

Pull request #1062 addresses the REQUIRED_IF issue for all of the "subarrayn_monthly_tilt" inputs

cpaulgilman, 2023-10-03:
Thanks, @sjanzou.

closed 2023-10-03 by cpaulgilman