NatLabRockies/ssc #1090: pvsamv1: non-linear beam self-shading derate not accounted for in annual/waterfall outputs

caseyzak24, 2023-11-16:
When you select standard shade mode where beam shading is treated nonlinearly, the loss due to beam self-shading is applied to dc power after the single-diode module model. But the annual_poa_shading_loss_percent output just seems to take the diff between poa_nom and poa_shaded, so it doesn't capture the dc-side beam derate. There doesn't seem to be another annual/waterfall output that does capture that loss, so it is effectively missing from a waterfall-type annual summary despite being the most important shading component for the most common shading scenario.

I noticed this when looking at the `develop` branch

cpaulgilman, 2023-11-16:
Some additional context:

Description of SAM's "non-linear self-shading" model: Deline, C.; Dobos, A.; Janzou, S.; Meydbrey, J.; Donoval, M. (2013). [A Simplified Model of Uniform Shading in Large Photovoltaic Arrays](http://www.sciencedirect.com/science/article/pii/S0038092X13002739). Solar Energy Vol 96, October 2013, pp 274-282. Draft Preprint ([PDF 1.3 MB](https://sam.nrel.gov/images/web_page_files/deline-self-shading-model-draft-2013.pdf))

The non-linear self shading model represents the effect of shadows as a reduction of the DC output and a reduction in the diffuse POA irradiance, which is consistent with what is shown in the PV loss diagram.

[image]

Showing a reduction in beam irradiance in the loss diagram would misrepresent how the non-linear self shading model works by suggesting that the POA input to the module model is less than the actual value.

The loss diagram does not break out the DC losses calculated inside the module model as discussed in https://github.com/NREL/ssc/issues/205. One way to quantify the loss is to run two simulations: One with self-shading enabled and one without.

The non-linear self shading factors are reported in the time series results as:

* Subarray n Self-shading non-linear DC factor
* Subarray n Self-shading non-linear ground diffuse irradiance factor
* Subarray n Self-shading non-linear sky diffuse irradiance factor

caseyzak24, 2023-11-17:
Paul,
Thanks for the breakdown. Respectfully, I still think the DC effect of beam shading should be broken out as a separate line item. To me, "module deviation from STC" implies deviations in magnitude of irradiance and temperature. And the dc effect on beam shading is already calculated separately as part of the shading step so it seems like it is strictly a reporting issue. For example, PVsyst reports this as "electrical effect". The name is a bit vague but once you understand what it is, it is clear that all aspects of shading phenomena are accounted for.

It seems like based on #205 an effort is already underway to have more individual losses broken out; if that is the case I'm happy to just have this duped or added to #205

janinefreeman, 2023-11-21:
Thanks @caseyzak24 ! I agree we will want to break out DC nonlinear shading loss out separately as part of #205  as well. I've added a comment to that issue to make sure this discussion is referenced, so closing this issue for now.

closed 2023-11-21 by janinefreeman