NatLabRockies/ssc #1111: Adjustment factors for Non-annual PV simulations

dguittet, 2023-12-12:
**Describe the bug**
When running a non-annual simulation where the weather data doesn't start at timestep 0, such as in https://github.com/NREL/pysam/blob/main/Examples/NonAnnualSimulation.ipynb

the adjustment factors are not properly aligned and result in a adjustment factor of 0, zeroing out all ac power as in the example above.

The length of the adjustment factors vector is the length of the weather file (810), but when accessing the adjustment factor, the timestep is used (3096) and since this is larger than the length, the default of 0 is returned.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the above notebook and see that the prediction line at the bottom is 0

**Expected behavior**
Adjustment factor should be aligned

janinefreeman, 2024-01-02:
@dguittet I would have expected the haf setup function to fail with the adjustment factors vector length of 810- my read of it is that it is supposed to require a complete 8760 (or subhourly) even for non-annual simulations, so indexing into hour of year would work even for non-annual simulations. I think we either need to (1) fix the checks so that it fails if non-annual arrays are passed in (not obvious to me why it didn't catch this), or (2) change the setup function to check that the length of the adj timeseries is equal to wf nrec, which requires the haf function knowing whether it's annual or non-annual....

dguittet, 2024-03-15:
@janinefreeman Apologies it took so long to get back to this.

So this issue of adjustment factors shows up when it's Non Annual and not starting at the beginning of the year. The adjustment factors are set up with `adjust_constant = 0`, not with "adjust_hourly" timeseries of adjustment factors. So then adjustment_factors runs `haf.setup(nrec, nyears)` which creates a vector of `nrec`:

https://github.com/NREL/ssc/blob/5010af992f298d600cbeddf4c5dd1fad17cf4b27/ssc/common.cpp#L1073

However in this case, the hour_of_year != idx and the call to `haf()` returns 0 instead of 1.

`if ( time < m_factors.size() ) return m_factors[time];`
https://github.com/NREL/ssc/blob/5010af992f298d600cbeddf4c5dd1fad17cf4b27/ssc/common.cpp#L1189

resulting in `ac` being 0.

I've set up a test showing this: https://github.com/NREL/ssc/pull/1113/commits/5010af992f298d600cbeddf4c5dd1fad17cf4b27

You can see

`p_ac[idx] = (ssc_number_t)(ac * haf(hour_of_year)); `

And the NonAnnualSummerStart test will fail, with ac = 0. Changing it to 

`haf(wdprov->annualSimulation() ? hour_of_year : idx)` fixes it.

I'm open to different ways to adjust for the indexing issue

janinefreeman, 2024-03-27:
Thanks for the additional details @dguittet ! I was looking at the wrong part of the setup function I think. Seems to me that your approach makes the most sense!

dguittet, 2024-03-27:
@janinefreeman Thanks for the review. I've switched the code so that the tests should pass. After they pass, I'll merge

PR #1113 (2024-03-28): fix adjustment factor use for pvwatts
fix #1111
closed 2024-03-28 by dguittet