NatLabRockies/ssc #1142: Reversible degradation for battery is not enabled

dguittet, 2024-02-29:
**Describe the bug**
The battery lifetime models currently do not allow for reversible thermal degradation due to this line: https://github.com/NREL/ssc/blame/936dd955c46799fe721b315576a395220b12efa9/shared/lib_battery_capacity.cpp#L523

The capacity (qmax_lifetime) can only be decreased because of the if statement. If the capacity_percent increases, the qmax_lifetime still cannot increase. This prevents the thermal reversible degradation from reversing.

**To Reproduce**
Use this SAM project file which has a battery that is not operating while it sits for a month at 0 deg then 60 deg.
[battery_reversible_fade.sam.zip](https://github.com/NREL/ssc/files/14452868/battery_reversible_fade.sam.zip)

The plot shows that as the temperature changes (black), the relative capacity also changes, recovering some capacity when the temperature increases (blue). 

The charge (red) and max charge (orange) however do not recover.
[image]

**Expected behavior**
The expected behavior is for the charge (red) and max charge (orange) to increase again.

[image]

brtietz, 2024-05-03:
@dguittet Would you consider https://github.com/NREL/SAM/issues/1291 related?

dguittet, 2024-05-03:
@brtietz It's kind of related. I've been seeing that issue as I've been working on this one, and it's due to that "Battery maximum charge at temperature (Ah)" being meaningless for NMC or LMO/LTO degradation models because it's always 100:

https://github.com/NREL/ssc/blob/936dd955c46799fe721b315576a395220b12efa9/shared/lib_battery.cpp#L139

closed 2024-06-05 by dguittet