NatLabRockies/ssc #1151: Battery retail rates fails with hourly dispatch and TOU demand charges

brtietz, 2024-03-19:
**Describe the bug**
It appears https://github.com/NREL/ssc/pull/1135/files added a new bug that shows up in the default SAM case. It looks like an uninitialized memory or other overrun issue. Turning off the demand charges solves the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a PV-Battery commercial case
2. Change the battery dispatch to retail rates
3. Run the simulation
4. See error

**Expected behavior**
Retail rates dispatch should run this configuration of rates.

**Screenshots**
Example output:

[image]

**Desktop (please complete the following information):**
 - Version 2023.12.17r1

cpaulgilman, 2024-03-20:
My review of #1135 missed this. Perhaps an additional test before approving PRs for for battery dispatch and utility rate code changes should be to run the "default" case for each battery dispatch option to check for simulation errors.

brtietz, 2024-04-25:
This was fixed in https://github.com/NREL/ssc/pull/1152

closed 2024-04-25 by brtietz