NatLabRockies/ssc #1171: Zeroes in output

ilozka, 2024-06-03:
Battery analysis using a 3-battery configuration gives acceptable results.
But the same analysis for a 2-battery configuration gives lots of zeroes in the batt_annual_charge_energy, batt_annual_discharge_energy and other output variables. LK files attached.

The 3battery script gives the following result:
[ 6694.57, 6681.41, 6669.19, 6656.1, 6644.09, 6632.22, 6621.02, 6610.46, 6599.98, 6589.87, 6597.16, 6583.9, 6571.53, 6559.71, 6548.62, 6537.52, 6524.22, 6511.54, 6498.89, 6482.25, 6494.04, 6476.61, 6459.39, 6443.02, 6427.03 ]
[ 6563.07, 6541.77, 6527.72, 6514.61, 6501.17, 6488.46, 6475.46, 6462.4, 6448.98, 6436.68, 6457.17, 6442.74, 6427.34, 6413.91, 6400.17, 6387.26, 6371.41, 6354.13, 6333.36, 6310.64, 6342.77, 6321.61, 6303.17, 6283.93, 6261.54 ]

While the 2battery script returns the values below - not sure if the numbers are correct.
Thank you for checking this out.

[ 25.814, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6033.01, 5874.5, 5742.03, 5619.59, 5486.05 ]
[ 29.0689, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3.54411, 0, 0, 0, 0, 0, 0, 0, 0, 0, 5789.5, 5612.32, 5473.48, 5341.57, 5196.36 ]

[3batt.lk.txt](https://github.com/user-attachments/files/15530698/3batt.lk.txt)
[2batt.lk.txt](https://github.com/user-attachments/files/15530699/2batt.lk.txt)

cpaulgilman, 2024-06-05:
@ilozka Could you provide some context so we can understand the project you are trying to model and the results you are exploring? It would also be helpful if you could attach the .sam file you used to generate these scripts.

Thanks,
Paul.

ilozka, 2024-06-14:
We're calling the battery simulation directly based on the battery selections in our application, so I can't provide any .sam file for this scenario - wondering why the 2-battery config returns zeroes for batt_annual_charge_energy and batt_annual_discharge_energy - if you compare those two LK files, you can see that the input differences are minimal, however, the result is different - why batteries are not charged and discharged if we have 2 of them instead of 3?

brtietz, 2025-03-17:
@dguittet this might be more in your wheelhouse than mine, since it appears to be a numerical error in the NMC lifetime model code. Here's what I've found so far:

- The scripts above need to be run in the debug version of the sdktool
- In hour 95, the capacity crashes to a large negative number based on a large nmc_li_neg->dq_relative_li1
- This appears to be caused by an extremely small nmc_li_neg->dq_relative_li1 in a prior step, based on a very small b1 which starts out a 2.48e-18 in hour 23

Is there something weird with an early cycle that causes this? Do you have thoughts on how to prevent it?

closed 2025-04-07 by sjanzou