NatLabRockies/ssc #1180: Translating from Power to Energy in Sunrise/Sunset Hours

janinefreeman, 2024-07-10:
Currently in SAM, for sunrise/sunset hours, power is translated to energy for the entire hour, not just the portion of the hour where the sun is actually up. This was possibly an ok assumption when most irradiance data was hour-averaged, but using instantaneous data, this may be more inaccurate. There's very little documentation from other pv tools as far as what they are assuming to go from power to energy for these hours.

janinefreeman, 2024-10-09:
@mikofski and @adriesse can I interpret your thumbs up emojis as agreement that we should change the assumption to assume power generation just for the portion of the hour where the sun is actually up? :)

mjprilliman, 2024-10-09:
https://github.com/williamhobbs would be great if you could weigh in as well.

adriesse, 2024-10-09:
@janinefreeman I was mostly just encouraging the initiative.  For a specific solution I would probably need a more detailed explanation or a formula.

williamhobbs, 2024-10-09:
@mjprilliman and @janinefreeman, remind me, is this a fix for the suspected root cause of the issue we saw in the subhourly correction work for PVSC?

mikofski, 2024-10-09:
I agree with Anton, I think context is useful.

if the sunrise time is known, then I think the average GHI energy during the interval containing sunrise would yield a lower incident flux if divided evenly over the entire interval versus assuming zero flux prior to sunrise then dividing the GHI energy over the remainder of the interval. This larger incident flux might have a different conversion efficiency. Also other effects like IAM and tracker position would also differ if centered at the middle of the entire interval versus centered on the remainder of the interval after sunrise.

However, I have a feeling this wasn’t the question you’re asking.

now that you have an estimate of the output during the remainder of the interval after sunrise, should this be assigned to the entire interval or not?

in my humble opinion, yes, we should keep the intervals the same and make sure that we only use units of energy which doesn’t say _exactly when_ that energy was produced only what the sum of all output over the interval was.

why? because this makes aligning the output with the input easier. The input and output are easier to work with if the intervals are the same size

cwhanse, 2024-10-09:
I think it would awkward if the energy output had irregular time steps at the beginning and end of each day.

janinefreeman, 2024-10-10:
Thanks all for your thoughts! @williamhobbs yes, this is one of the issues that arose out of this work.

Further context- SAM calculates the sun position (and therefore all related incidence angles) at a given point during a time step. For hourly data, that's usually at the 30 min mark. For sunrise and sunset hours, though, the convention is to use halfway between (sunrise/end of hour) or (beginning of hour/sunset) for the timestamp of the sun position calculation (unless the exact minute to use is specified in the weather file). However, when we then translate the instantaneous _power_ calculation to _energy_, this is problematic for the sunrise/sunset hours.

A worst case (totally fictitious) example- sunrise is at 6:50AM in an hourly simulation. We calculate instantaneous power at 6:55AM to be 10 kW. Currently, that would be translated into 10 kWh energy for the 6-7am timestep, when in reality it is probably closer to 1.7 kWh for that timestep. 

The proposed fix we discussed at our SAM meeting yesterday is to scale the **power** value to be the _average power over the timestep_, so in our example, we would assume 10 min of 10 kW and 50 min of 0 kW, and assign a **power** value of 1.7 kW for the 6-7am timestep. This will maintain simplicity in subsequent energy summations (not having to treat sunrise/sunset timesteps differently than every other timestep), and we think should compare better to measured data, assuming that measured data is usually an _average_ value over some time period (or measured data is actually energy, in which case the user is performing a unit conversion anyway).

The same principle would apply regardless of timestep, just using hourly data as the worst-case example.

The reason we're trying to maintain this in a power calculation, versus switching to energy, is that we need power values for downstream calculations in financial models (best example is demand charges for behind-the-meter systems offsetting a load with PV).

I think our proposed approach aligns with the comments I'm reading here, but appreciate any further input from our impromptu think tank 😄

cwhanse, 2024-10-10:
> we need power values for downstream calculations in financial models (best example is demand charges for behind-the-meter systems offsetting a load with PV).

Demand charges are frequently based on shorter intervals, e.g., SDGE uses 15 minutes.

I think it's better to leave power "unaveraged", that is, publish the point value that is estimated for each interval of time, along with the energy accumulated for each interval of time. With the average power, energy pair, the two numbers will always be the same (1.67kW, 1.67kWhr) for the example above. In contrast, 10kW power and 1.67kWhr energy provides more information, and enough to reconstruct what happened in the simulation.

adriesse, 2024-10-10:
Would there be a difference between the calculations for period-averaged weather data vs. instantaneous?

adriesse, 2024-10-11:
> A worst case (totally fictitious) example- sunrise is at 6:50AM in an hourly simulation. We calculate instantaneous power at 6:55AM to be 10 kW. Currently, that would be translated into 10 kWh energy for the 6-7am timestep, when in reality it is probably closer to 1.7 kWh for that timestep.

This case arises only when you have hourly average irradiance input, right?  If you had had instantaneous values you would probably have had zero irradiance at 6:30 as input for the hour.  So if the average irradiance for the sunrise hour produced 10 kW, then 10 kWh seems like the right energy.

cwhanse, 2024-10-11:
> that requires massive refactoring of all of our financial models,

Understood.

mjprilliman, 2024-10-21:
> This case arises only when you have hourly average irradiance input, right? If you had had instantaneous values you would probably have had zero irradiance at 6:30 as input for the hour. So if the [...]

In thinking about this some more, I agree it makes sense to only look at instances where the irradiance is average, as instantaneous irradiance at a given time should be calculated for solar position at that time regardless of sunrise/sunset time. Currently, only hourly data input to SAM can be provided without a specified minute, and as we don't currently have a input flag to specify subhourly data as instantaneous or average it feels unsafe to assume which the user has provided. I also agree with the point that the averaged irradiance should already account for the time with no irradiance (i.e. before sunrise / after sunset) in the hourly averaged values. So does this mean that no corrections should be made? Is it more a question of where to assume the solar position is in sunrise/sunset hours when not specified?