NatLabRockies/ssc #1265: Clarify depthThreshold default value

cwhanse, 2024-12-12:
`depthThreshold` is described as https://github.com/NREL/ssc/blob/7e7c049e53f073e9a77b7eaa1f64097c3342b725/shared/lib_snowmodel.h#L52

The default value of depthThreshold appears to be 1.0 cm. https://github.com/NREL/ssc/blob/7e7c049e53f073e9a77b7eaa1f64097c3342b725/shared/lib_snowmodel.cpp#L67

The [reference report](https://www.nrel.gov/docs/fy17osti/[68705.pdf](https://www.nrel.gov/docs/fy17osti/68705.pdf)) says: "...when calculating the snow coverage at the beginning of each time step, we included an additional check for zero snow depth at that time. We assume that if the snow depth at that time is zero, then the coverage percentage on the PV array should also be zero. "

Should the default be 0cm?

janinefreeman, 2024-12-19:
Hi @cwhanse ! The default is 1 cm due to the high frequency of snow depth measurement devices reading small amounts of snow > 0cm, even when there is no snow on the ground (July in Golden, for example). In the snow model implementation, "zero" is anything less than the depth threshold, per:

https://github.com/NREL/ssc/blob/7e7c049e53f073e9a77b7eaa1f64097c3342b725/shared/lib_snowmodel.cpp#L126

Does that help?

cwhanse, 2024-12-19:
Yes, thanks @janinefreeman. We'll include that note in the pvlib code.

closed 2024-12-19 by janinefreeman