NatLabRockies/ssc #1283: Windpower module applies HAF twice

ppinchuk, 2025-02-24:
I believe the second call to `haf.setup` here may be applying the hourly adjustment factors twice:

https://github.com/NREL/ssc/blob/82d9c9ebb98b556018e02ffa8464f53528be31d4/ssc/cmod_windpower.cpp#L569-L570

I noticed this when running PySAM 5+. I am inputting an hourly adjustment factor of 17% for a particular timestep, but SAM gives back a power generation value that is reduced by 31.11% (= 1 - 0.83 * 0.83).

Happy to provide more specific details if needed.

cpaulgilman, 2025-02-24:
@ppinchuk This appears to be working correctly in the latest SAM 2024.12.12, which is equivalent to PySAM 6.0.1.

Could you share a .sam file or minimum reproducible Python script that demonstrates this result?

ppinchuk, 2025-02-24:
Sure, below is a modified version of the [windpower example directly from PySAM](https://github.com/NREL/pysam/tree/main/Examples/Windpower_Example) (P.S. the one in the repo doesn't run out-of-the-box because the [`constant`](https://github.com/NREL/pysam/blob/994b82c05d9b1a8f28bb4c762df7eb5b21756736/Examples/Windpower_Example/windpower-inputs.json#L39) key in the windpower-inputs JSON was renamed to `adjust_constant`).

```python
import json
from pathlib import Path

import PySAM
import PySAM.Windpower as wp

print(f"Running with PySAM version {PySAM.__version__}")

wind = wp.new()
working_dir = str(Path(__file__).parent)

with open(working_dir + '/windpower-inputs.json', 'r') as file:
    sam_data = json.load(file)
    for k, v in sam_data.items():
        if k != 'number_inputs':
            wind.value(k, v)

wind.value(	'wind_resource_filename', working_dir + '/wind-resource-file.csv')
wind.value('wind_resource_shear', 0.14)

wind.execute()

baseline_gen = wind.Outputs.gen[0:10]
print('First 10 gen values (baseline) = ', baseline_gen)

haf = [0] * 8760
haf[0:10] = [5, 10, 15, 20, 25, 30, 35, 40, 45, 50]

wind.value('adjust_hourly', haf)
wind.value('adjust_en_hourly', 1)

wind.execute()

adjusted_gen = wind.Outputs.gen[0:10]
print('First 10 gen values (with HAF) = ', adjusted_gen)

losses = [(1 - a/b) * 100 for a, b in zip(adjusted_gen, baseline_gen)]
print('Expected HAF:', [round(float(a), 2) for a in haf[0:10]])
print('Output HAF:', [round(a, 2) for a in losses])
```

Running this script give me the following output:

```
Running with PySAM version 6.0.1
First 10 gen values (baseline) =  (322.9265538249852, 347.98610853955586, 259.93752706734045, 222.29397194684452, 304.6571663197476, 335.7182054088018, 142.40439289046776, 135.0457597477396, 132.5973603599542, 175.10135185809602)
First 10 gen values (with HAF) =  (291.4412148270491, 281.86874791704025, 187.80486330615346, 142.2681420459805, 171.369656054858, 164.50192065031285, 60.16585599622263, 48.61647350918625, 40.11070150888615, 43.775337964524006)
Expected HAF: [5.0, 10.0, 15.0, 20.0, 25.0, 30.0, 35.0, 40.0, 45.0, 50.0]
Output HAF: [9.75, 19.0, 27.75, 36.0, 43.75, 51.0, 57.75, 64.0, 69.75, 75.0]
```

The expected behavior would be for "Output HAF" to match "Expected HAF". Right now, you can easily verify that it's giving back `(1 - in_haf * in_haf)`

[windpower-inputs.json](https://github.com/user-attachments/files/18950041/windpower-inputs.json)

[wind-resource-file.csv](https://github.com/user-attachments/files/18950043/wind-resource-file.csv)

janinefreeman, 2025-02-25:
@cpaulgilman I can replicate in desktop 2024.12.12 . @ppinchuk I think you're right about the second call to setup. We're planning a patch or small new release coming up soon, I'll add this issue to that.

cpaulgilman, 2025-02-25:
@janinefreeman Yes, I see the problem and have it fixed locally. I'll make a pull request after some further testing.

PR #1284 (2025-02-26): Fix wind adjustment factor double-counting
Fix #1283

To test, run a default Wind / No Financial case with and without time series adjustment factors. The difference between the two runs should be the adjustment factor applied once, not twice as reported in #1283.

[image]

The adjustment factors should work correctly for the following combinations:

* Hourly weather file with hourly losses
* Subhourly weather file with subhourly losses
* Weibull distribution with hourly losses
* Wind resource probability table with hourly losses
* Constant loss wake model option with hourly or subhourly simul [... cut]
closed 2025-02-26 by cpaulgilman