NatLabRockies/ssc #1292: Move more cmods to the attic

brtietz, 2025-03-05:
Code coverage indicated a number of cmods that could stand to be moved to the attic. The list so far is:

-cmod_fossilgen.cpp
-cmod_utilityrate(1-4).cpp

Flesh out this list and move these to the attic for the next release

janinefreeman, 2025-03-06:
I don't see any other obvious ones to move (to me), but would appreciate if @tyneises  would check that there aren't any old/unused csp compute modules hanging around. Thanks!

brtietz, 2025-03-14:
Partially solved with https://github.com/NREL/ssc/pull/1297. I believe there is a CSP attic PR coming, so leaving this open for now as a reminder.

tyneises, 2025-03-21:
CSP clean up here https://github.com/NREL/ssc/pull/1299 and here https://github.com/NREL/SAM/pull/2037

closed 2025-04-02 by brtietz